jlempen/Surface-Laptop-3-OpenCore #1: [Question] What's the point of CPUFriend? (& other things)

ConwayTech-Dev, 2024-10-01:
Hey! First of all, the work you've been doing for BigSurface and for various Hackintosh projects in general is very admirable. Thanks for that!

**(Note: this issue is _really_ long. I have an unfortunate habit of making GitHub issues/PRs far too long, and this issue unfortunately follows that pattern, covering far too much for one issue.)**

Anyway, I've been using an SL3 (i7, 16GB RAM, 13-inch) Hackintosh as one of my daily drivers for about 7 months now and have never used CPUFriend (I know almost nothing about it, I've only briefly peeked at the code). The i7-1065G7 in the SL3 is, as I'm sure you know, not natively supported by macOS. (The equivalent SMBIOS, MacBookAir9,1, which I use on my SL3, has a i7-1060NG7.) I've heard that CPUFriend is designed to help CPU power management for Hackintoshes that aren't natively supported, so what exactly would it do if using it with SL3?

I ask because I've had two frustrating issues with my SL3 Hackintosh that are almost certainly related to the CPU. First of all, the CPU gets really warm (70c-90c degrees, which is extremely abnormal and unhealthy (measured via [Stats](https://github.com/exelban/stats)) and the entire computer throttles like crazy. It becomes completely unusable, with the screen becoming almost unresponsive. This is natural due to the abnormally high temperature, but do you think CPUFriend could help against overheating? (I usually run into this issue when using the laptop on something like a couch; additionally, the last time I've cleaned the fans was probably over a year ago, so I'll definitely try clean the laptop when I have time.) Second of all, when the computer is below ~10% battery, its becomes exactly like what I previously described — macOS (particularly animations) stutters and takes forever to do simple tasks. I've tested this on various occasions, and it's always the same. Most recently, I ran the battery down to 10% and then shut off the computer. A day later, I shut on the computer again and noticed the same struggles (even bootup was slow, and text in verbose was moving ridiculously slowly). Do note that this issue only happens if the computer is turned on while below 10%. When the laptop is above 10% and goes below 10% after use, performance is always solid like normal (even at 1%). Also, my battery capacity is ~83%. Battery doesn't help long, and I'm hoping CPUFriend could fix that.

I plan on testing some other things with the SL3 soon — I'm coming back from hiatus in the project and will update BigSurface 6.5 to your modified version among other things. Is there any advantage to using the MacBookPro16,2 SMBIOS instead of the MacBookAir9,1 SMBIOS? I know the 16,2 has a i7-1068NG7 — what difference does SMBIOS make performance-wise? Also, completely unrelated, but what HiDPI resolution do you use? I'm rather unhappy with mine, but it's the best BetterDisplay will let me do.

Last, BigSurface is currently behind on various things — dependencies need to be updated and a new release hasn't been published in a while. I assume @Xiashangning is taking a break for now and won't update the kext for a while, so do you plan to take over the kext while they are away?

Thank you in advance for your help, and for reading my long GH issue! I'm happy to help test anything or do anything SL3-related in the future and am grateful that you're providing useful resources for the Hackintosh community. Hopefully you can help me out as an experienced Hackintosher.

ConwayTech-Dev, 2024-10-10:
I'm closing this issue for a couple of reasons:
- It's far too long
- I've fixed many of my issues in the above issue and made significant progress in this project
- I'm creating another issue that's more up-to-date

jlempen, 2024-10-11:
Hey ConwayTech-Dev,
Sorry for answering only today, I'm extremely busy at work right now.
Your post isn't too long at all :-)

> I've fixed many of my issues in the above issue and made significant progress in this project

Please contribute back to this repo if you fixed important things so everyone could benefit from your findings, your pull requests are welcome!

But just to quickly add my 2 cents:

Your CPU shouldn't be reaching high temperatures like that, my i5 13" and i7 15" SL3s don't heat up like crazy even when I compile stuff with Xcode... Sometimes I'll hear the fans spin up for a few minutes, but not while doing trivial stuff like Firefox / Mail / VLC / Plex.

Regarding the SMBIOS, from what I understand, MacBook Pros usually came with more powerful CPUs than MacBook Airs. The default values from the CPUFriendFriend CLI suggest that the frequency vectors used by default by Apple for the MacBook Airs are a bit more oriented towards performance to compensate for the weaker CPUs and the MacBook Pros are a tad more oriented towards power saving to increase the runtime on battery.
Changing the SMBIOS should not make a big difference in temperature or performance (if any at all), but using CPUFriend and the CPUFriendFriend tool allows you to define your own frequency vectors for the CPU and thus you could achieve lower temperatures at the expense of lower performance as well. Feel free to play around with CPUFriendFriend, I've tried the lower or lowest settings everywhere once and the OS still runs quite OK, but Apps take ages to start :-)
I personally use the default vectors from my SMBIOS (CPUFriend disabled in my config.plist), which works well for me, providing me with a cool CPU and balanced performance with around 5 hours runtime on battery (i7 SL3 15"). I also set "Low Power Mode" to "Always" in the System Settings, because I don't need much more performance than that.

> Also, completely unrelated, but what HiDPI resolution do you use? I'm rather unhappy with mine, but it's the best BetterDisplay will let me do.

I don't have the machine on hand here, but BetterDisplay should give you literally dozens of resolutions for your SL3. There's also a few options to check in the settings so that "better" resolutions are added to the native macOS Display Settings: in "Displays" select your "Built-In Display", then under "System Configuration and Flexible Scaling" enable "Edit the system configuration of this display model". Then restart and you should have nice defaults directly in macOS's Display Settings.

> I assume @Xiashangning is taking a break for now and won't update the kext for a while, so do you plan to take over the kext while they are away?

Xiashangning is still active, but very infrequently.
As I don't have the fantastic programming skills he has (I'm no programmer at all, but I understand code quite well as I used to write code quite a lot in another life a very long time ago), I can't really maintain the BigSurface project other than answering a few questions and fixing trivial bugs. So don't expect much from Xiashangning or myself, but I'll try my best to make sure that IceLake Surface devices will still work in macOS Sequoia (or whatever version will be the last one supporting Intel CPUs) with the functionality they have now such as working battery readings, native trackpad gestures and hibernation. Don't expect any new stuff like double battery support or keyboard and trackpad support after detaching and reattaching the clipboard (keyboard part) of the SB3 line, for instance.

amethyst8118, 2025-03-09:
Check this out, it should solve the throttling issue. @ConwayTech-Dev 

https://github.com/Xiashangning/BigSurface/issues/127#issuecomment-2271896057

jlempen, 2025-05-24:
Closed due to no activity

closed 2025-05-24 by jlempen

ConwayTech-Dev, 2025-05-26:
Hey @jlempen,
I'm so sorry! I completely forgot about responding to your message — I've been busy, and I think I must have glossed over the GitHub notification for your reply until now. Completely my fault, sorry for the inconvenience.

Unfortunately, I must report that I haven't been working on improving the Hackintosh aspect of my SL3 much. That's because it's in a state where other than a couple of minor issues, everything is going great. So I've just been enjoying the macOS experience on the great Surface hardware. (I do have an actual macOS laptop, though, so I use the SL3 less — maybe for a few hours per week.)

Anyway, I thought I'd quickly give an general update on all parts of my SL3 experience, since I've been a SL3 Hackintosh user for over a year now.

A while ago, I ended up cleaning the SL3's fans, and perhaps as a result, the CPU throttling issue basically never happens when I'm using the computer on a table and giving it decent airflow. Furthermore, the temperatures are fine. (I have an i7.)

That's great, but I have noticed that when the computer is resting on a couch or has blocked airflow, the temperatures become horrendous and the CPU throttling does indeed kick in. In these scenarios, the CPU idles at 70-100 Celsius, and the CPU throttles quite a bit. Now, I'm sure that disabling BD-PROCHOT would fix my unusable CPU, but I'd actually rather not install it. The temperatures are already very dangerous, and it's better to be safe than sorry. Instead, I'm planning on just giving the computer more airflow and using it on a table when possible, which also seems to be a fix to the issue.

(I did test CPUFriend, but it didn't really do anything. I'm guessing I don't know how to configure it. I still have it enabled in my EFI, though.)

My Sonoma AirportItlwm Wi-Fi is frustrating. It works, but it's extremely intermittent. I've been using the laptop for over a year, and I've noticed that very brief Wi-Fi disconnects happen a few times every hour. Whilst the disconnects are very brief, they become very annoying. Honestly, though, I think that's just the price to pay for a Hackintosh. It's also a shame that AirportItlwm isn't available for Sequoia; looks like that project is dead.

If I recall correctly, I use 1440x960 HiDPI in BetterDisplay. I compared it to a real M1 MacBook Air and that seemed to be the best resolution. It's no longer an issue.

HDMI is not working, as someone else mentioned. I tried using Microsoft's Surface Dock with a MiniDP cable. I also tried HDMI via a USB-C dongle. In both scenarios, the computer shuts down as soon as you connect the cable. If you connect the cable before boot, then try to turn on the laptop, it just won't detect the display and nothing will happen on the external screen.

The battery life is absolutely horrible, so I'm considering setting Low Power Mode to be permanently on. Then again, my battery health is reported to be 82% by Stats, so that's probably why.

Boot times are also pretty bad when compared to the Windows boot time for this laptop. Any ideas there? We could try getting rid of unnecessary kexts, but I'm not entirely sure that there are any.

Anyway, that's my update on the SL3 Hackintosh experience. Thank you so much for maintaining this great repository and helping out other SL3 users!