jlempen/Surface-Laptop-3-OpenCore #3: Downgrading UEFI using the method in the tutorial in the latest UFEI fails

hoooxl, 2025-01-05:
Downgrading UEFI using the method in the tutorial in the latest UFEI fails
And the touchpad is not available
I have strictly followed the tutorial method for downgrading, and a BIOS update progress bar has appeared
The current UEFI version number is: system UEFI 19.101.140

jlempen, 2025-01-07:
@hoooxl 
After the downgrade, the current UEFI version should be 13.101.140.0.

Did you reboot into Windows at some point after the UEFI downgrade?

At the moment, it is not possible to have Windows installed at the same time as macOS, as the SL3 will immediately update to the latest firmware again when you reboot into Windows, even if you turn off your WiFi or disable automatic Windows Updates. I haven't found a way to prevent this behaviour yet.

hoooxl, 2025-01-12:
After downgrading, it didn't reboot into Windows. I found that the touchpad didn't work after booting into Linux. When I went back to the UEFI interface, I found that the version number hadn't decreased. This is a failed case. In fact, my touchpad had been physically damaged and has been replaced with a new one. It's not recommended to continue trying.

jlempen, 2025-01-12:
Well there's something fishy going on, because in my own experience, the trackpad works in Linux and Windows with any version of the UEFI BIOS.

Even if your trackpad was physically replaced, it should still be detected as the same hardware ID. Unless it was replaced with a trackpad from another device (Surface Book 3 or Surface Laptop Go).

I would recommend rebooting into Linux and repeat the whole process. Once the SL3 starts flashing the firmware, the process takes somewhere around 5 to 10 minutes to finish.

https://github.com/jlempen/Surface-Laptop-3-OpenCore?tab=readme-ov-file#downgrading-the-uefi-firmware

hoooxl, 2025-01-13:
I dare not continue operating now
Afraid of damaging my touchpad again
Could you please help me check if there are any issues with my operation

[image]
[image]

jlempen, 2025-01-14:
@hoooxl Everything's looking good so far :-)

hoooxl, 2025-01-14:
I made another attempt at the risk of failure, yet the UEFI downgrade still didn't succeed. Luckily, my touchpad remained intact physically this time. I've repeatedly verified that there were no mistakes in my operations. The sole distinction is that I installed Linux onto the physical hard disk. I'm not certain whether this has made a difference. I've already taken a photo of the UEFI interface after the downgrade.

[image]
[image]

jlempen, 2025-01-14:
There's definately something wrong.
When your SL3 is flashing the firmware such as in the first screenshot you sent today, does it flash several files (i.e. you see several loading bars with different colours one after the other) or only the green one?
Wait a second, do you use my latest EFI folder in your SL3? Because if you only download my EFI and insert your SMBIOS information in it and start your hackintosh, you should have a working trackpad and keyboard, but the trackpad will skip every few seconds. That's the first thing I would check for.
I admit I haven't tested the firmware downgrade on the SL3 with the Alcantara keyboard deck, but it's working successfully on the black aluminum SL3 13-Inch and on the aluminum 15-Inch version.

jlempen, 2025-01-14:
There's actually only one firmware download for all SL3 devices at Micro$oft, so the SL3 model shouldn't make a difference.
What's the output in Linux once you execute the command in the second screenshot of your comment from yesterday?
https://github.com/jlempen/Surface-Laptop-3-OpenCore/issues/3#issuecomment-2587633395

hoooxl, 2025-01-14:
Two colors appear: green and gray

[image]

jlempen, 2025-01-14:
The problem is actually right in front of your eyes :-)

[image]

The lines I highlighted in green are perfectly normal, they are the ones I'm talking about in my instructions.

The lines I highlighted in red are real errors:
"failed to write firmware: /run/media/root/1F6B-8E85 does not have sufficient space"
Which means that the EFI partition has not enough space available to write all the firmware files...
And thus it flashes only the smaller firmware files and not all of them.

According to the output in your screenshot, you have to make room for 28.7 MB, 23,7 MB and 30 MB -> around 100 MB on the EFI partition in order to install and flash the remaining firmware.

There's a few options, for instance you might temporarily copy the Microsoft folder from your EFI partition to your desktop and delete it in your EFI folder. Once all the firmware files have been flashed, you can delete the Manjaro folder from your EFI partition, it contains the firmware capsule files which are not needed anymore once the firmware files have been flashed to the UEFI.

hoooxl, 2025-01-15:
I'm sorry to bother you again.I have already cleaned the EFI partition and deleted the files under the fwup folder.Then I went back to Linux and executed the command. Is the output result normal?

[image]

jlempen, 2025-01-15:
@hoooxl Yes, that looks just like on my SL3 :-) Go ahead and reboot, it will flash the firmware for 8 devices. Be patient, as the device will reboot several times and sometimes it may take up to 30 seconds before something is happening.
Please keep me posted on how it goes.

hoooxl, 2025-01-15:
[image]

The downgrade has been successful. Thank you very much.

jlempen, 2025-01-15:
Great news, thanks for the feedback! Have fun with macOS on your Surface Laptop 3!

jlempen, 2025-01-16:
Closing this as it appears to be resolved.