jlempen/Surface-Laptop-3-OpenCore #4: Some problems that occur after hibernation and wake-up

hoooxl, 2025-01-17:
After resuming from hibernation, issues occur, including incorrect time zone settings and a buzzing sound coming from the speakers.

https://github.com/[EternalTimes/LocalTime-Toggle](https://github.com/EternalTimes/LocalTime-Toggle)
This can resolve the time - zone issue after wake - up. However, the lock screen still doesn't function properly, while it returns to normal after unlocking.

There is a hissing sound in the speaker, which can only be eliminated after playing music. I'm wondering if there is a solution.

hoooxl, 2025-01-17:
The Wi-Fi and Bluetooth issues have actually been resolved under the macOS 15.2 (24C101) system. 
https://bbs.pcbeta.com/viewthread-2030386-1-2.html

However, when the laptop lid is opened after it goes into hibernation, the computer cannot be awakened, and the power button needs to be pressed instead.

[image]

jlempen, 2025-01-17:
> There is a hissing sound in the speaker, which can only be eliminated after playing music. I'm wondering if there is a solution.

This is "normal", it happens on all Surface devices (apart from the Surface Go 2) running macOS. It's because there's only a single ALC layout-ID for all Surface devices in the [AppleALC](https://github.com/acidanthera/AppleALC) repo and unfortunately it has this bug. It should be possible to fix it, but most pretty much normal human beings like myself are absolutely unable to understand AppleALC source code, let alone fix a buggy codecs... Feel free to have a look at it, it's not for the faint of heart :-)

Here's the basics: https://github.com/acidanthera/AppleALC/wiki/Adding-codec-support

And here's the harder stuff: https://github.com/5T33Z0/AppleALC-Guides

jlempen, 2025-01-17:
I have never heard of or experienced timezone errors on wake, what's the symptom?

Bluetooth issues have already been addressed and fixed in my repo: https://github.com/jlempen/Surface-Laptop-3-OpenCore#fix-broken-bluetooth-on-wake-from-hibernation

What are the symptoms of your wifi issues?

> However, when the laptop lid is opened after it goes into hibernation, the computer cannot be awakened, and the power button needs to be pressed instead.

Yes, that's an expected outcome if you interrupt the hibernation process. It's already covered in my repo:

"Keep in mind that once the Surface Laptop 3 hibernates, you need to let it hibernate for a couple of minutes before waking it up. Failing to do so will disrupt hibernation and the device will hang. You'll have to turn it off forcefully by pressing on the power button for 10 seconds. Sometimes you even need to press the power button for up to 20 seconds to restart the laptop."

hoooxl, 2025-01-18:
1: WIFI and Bluetooth are completely normal in macOS 15.2. There is a tutorial link above.
2: Under macOS 15.2, I woke up after completely sleeping. I should wake up when I open the lid normally, but I need to press the power button.
3: Regarding the issue of incorrect time after waking up, in my case the time after waking up will be 8 hours longer than the normal time, but the restart time is normal.

jlempen, 2025-01-22:
> 1: WIFI and Bluetooth are completely normal in macOS 15.2. There is a tutorial link above.

Thanks for the tip. It's kind of hacky though, I prefer waiting until the [OpenIntelWireless](https://github.com/OpenIntelWireless/itlwm) project releases an AirportItlwm.kext for Sequoia. That's one of the reasons I recommend staying on macOS Ventura for the time being.

> 1: WIFI and Bluetooth are completely normal in macOS 15.2. There is a tutorial link above. 2: Under macOS 15.2, I woke up after completely sleeping. I should wake up when I open the lid normally, [...]

When you hibernate your laptop, it saves the RAM to the SSD and shuts down completely, that's by design. Thus you need to press the power button to wake it up again.

> 3: Regarding the issue of incorrect time after waking up, in my case the time after waking up will be 8 hours longer than the normal time, but the restart time is normal.

As I have never experienced this bug on any of my hackintosh laptops, I won't investigate further.

hoooxl, 2025-01-22:
Regarding the issue of incorrect time after waking up, for me, the time after waking up is 8 hours longer than usual, but the restart time is normal.
I don't know what your time zone is, I am in the China time zone, which is +8, and I just woke up after a long sleep, which is 8 hours more, I think it's related to the time zone
The surface BIOS is different from that of a normal laptop, which should be a compatibility issue

hoooxl, 2025-01-23:
实际上我早就设置了自动更新时区，但是这个不起作用
 

小何
***@***.***

  

  
macOS should update the time automatically on wakeup when the "Set time and date automatically" option in System Settings -&gt; General -&gt; Date &amp; Time is enabled. Perhaps it needs a moment to update after hibernation.
 
&mdash;

alectprasad, 2025-04-10:
> Thanks for the tip. It's kind of hacky though, I prefer waiting until the [OpenIntelWireless](https://github.com/OpenIntelWireless/itlwm) project releases an AirportItlwm.kext for Sequoia. That's one of the reasons I recommend staying on macOS Ventura for the time being.

itlwm hasn't been updated in over a year. its dead.