jlempen/Surface-Laptop-3-OpenCore #5: Booting from internal disk does not work

alectprasad, 2025-03-04:
I can boot from the USB but the same EFI placed in internal disk fails to boot (no bootable OS found)

[image]

jlempen, 2025-03-04:
@alectprasad 
The first thing I would suggest is to clear the NVRAM by booting into your USB stick and pressing the space bar once you see the OpenCore picker. Then choose the option "ResetNvramEntry", then reboot and see if it helps. 

Also try to shutdown and restart the SL3, this might help.

Another thing to try is to enter the UEFI BIOS and make sure the "Internal disk" or "OpenCore" option is at the top of the Boot options list.

alectprasad, 2025-03-04:
i already did that. not sure what else to do

[image]

jlempen, 2025-03-04:
Try dragging the "Internal Storage" entry to the top and see what happens. Perhaps the "OpenCore" entry in the list is actually the one on the USB stick...

alectprasad, 2025-03-04:
that worked

closed 2025-03-04 by alectprasad

jlempen, 2025-03-04:
Great, have fun with your SL3 hackintosh :-)