jlempen/Surface-Laptop-3-OpenCore #8: is there any way to add fan controls?

alectprasad, 2025-04-09:
macs fan control is not detecting any fans and the cpu is idling at 60c

jlempen, 2025-04-09:
> macs fan control is not detecting any fans

Fan control is handled by the [VirtualSMC plugin](https://github.com/acidanthera/VirtualSMC) `SMCSuperIO.kext`. If it doesn't detect the fan of the SL3, there's not much else we can do.

> the cpu is idling at 60c

Is this happening on the 13-Inch or the 15-Inch SL3? I've never encountered this on the 15-Inch.

Try disabling BD PROCHOT which seems to kick in too early on your device. An UEFI extension such as [arter97's DisablePROCHOT](https://github.com/arter97/DisablePROCHOT) should do the trick.

jlempen, 2025-04-13:
@MirComputers
Good to know, thank you!

Are you talking about the 13-Inch or the 15-Inch? I have the impression that the 15-Inch has way less heat-related issues.

MirComputers, 2025-04-13:
sorry, 15 inch