jlempen/Surface-Laptop-3-OpenCore #11: Scrooling a bit Choppy / Laggy

fundave3, 2025-04-13:
Ive installed Ventura on my surface laptop 3
Its working but Ive reconized that sometimes the screen is a bit laggy.
That issue occurs only for seconds and then its smooth again.
But thats issue is occurring often.

Any idea how to solve ?

jlempen, 2025-04-19:
@fundave3 Are you sure it's the screen that's laggy and not the trackpad? Your issue sounds like the infamous "skipping/lagging trackpad" issue. [The only solution for this issue is to downgrade the firmware to the last known good version](https://github.com/jlempen/Surface-Laptop-3-OpenCore?tab=readme-ov-file#downgrading-the-uefi-firmware).

jlempen, 2025-05-24:
Closed due to no activity

closed 2025-05-24 by jlempen