jlempen/Surface-Laptop-3-OpenCore #12: Unable to connect to display.

jhmai1212, 2025-05-15:
I installed Ventura on my 1065G7 Surface Laptop 3, but when I connect a display, it crashes and restarts.

jlempen, 2025-05-19:
@jhmai1212 
As I use my SL3 only as a travel laptop, I don't have any external display to test this kind of stuff, sorry. I'm quite sure that the USB-C display output could be fixed, but someone would have to play around with the `DeviceProperties` to fix this issue.

jhmai1212, 2025-06-19:
"hda-gfx=onboard-1" can solve this problem.

MirComputers, 2025-06-19:
that is already in the config and fixes nothing..

jhmai1212, 2025-06-19:
> that is already in the config and fixes nothing..

Sorry, I made a mistake. It should be `No-hda-gfx = onboard-1`.

jlempen, 2025-06-19:
@MirComputers can you confirm that `No-hda-gfx` -> `onboard-1` fixes the external video over USB-C?

MirComputers, 2025-06-19:
nope, i took out the old value that was 020000 something and replaced type with "string" and set "onboard-1" but still: display sees input, goes dark, internal display goes dark aswell and fan start spinning, taking the usbc out keeps the internal dark and i have to manually shutdown

jlempen, 2025-06-19:
Well, both arguments actually already exist in my `config.plist` file:

`No-hda-gfx` with a data value of `00000000 00000000`

and

`hda-gfx` with a string value of `onboard-1`

I honestly don't know for sure what those arguments do, I think they are more related to audio than video, more specifically if audio is sent through HDMI to an external monitor or not...

jlempen, 2025-06-19:
I guess we'll need to do some more framebuffer patching to fix external monitor output through USB-C. This new section of the Dortania guide is a good starting point: https://dortania.github.io/OpenCore-Post-Install/gpu-patching/intel-patching/#terminology

jhmai1212, 2025-06-20:
[image]

[image]
Here's my setup

jlempen, 2025-06-20:
@jhmai1212 
Ah, now I see! Yes, you're talking about the `DeviceProperties` of the HDA codec at `PciRoot(0x0)/Pci(0x1F,0x3)`! I thought you were talking about the iGPU `DeviceProperties` at `PciRoot(0x0)/Pci(0x2,0x0)`.

All right, so in my `config.plist` there's actually both arguments as well:

[image]

So you're suggesting we have to comment out or removing the `hda-gfx` argument and replacing the `No-hda-gfx` with the string value `onboard-1`.

jlempen, 2025-06-20:
I think I'll borrow an external monitor from my neighbour and have a go at it.

jlempen, 2025-06-20:
OK, this is working for me on a huge Samsung monitor hooked up through a good USB-C cable. No HDMI adapter, although the monitor has two HDMI inputs as well.

I tried all resolutions up to 3840x2160 and it's working without a hitch.

The first time I hooked the monitor to the SL3, both the internal display and the Samsung would alternatively turn on and off. I only needed to unplug the monitor and replug it and now it's working fine.

Hotplugging is working fine and the external monitor turns on after a cold boot or a reboot.

I'm running Ventura 13.7.6 on a 15-Inch Surface Laptop 3.

jlempen, 2025-06-20:
Fixed with https://github.com/jlempen/Surface-Laptop-3-OpenCore/commit/5b1b5f582fb75ec5092219e428871bc24efa0357

closed 2025-06-20 by jlempen

MirComputers, 2025-07-05:
works at last!! thnks!!!