jlempen/Surface-Laptop-3-OpenCore #13: will this work for SL2?

alappatsj, 2025-05-19:
Hi,

will this work for Surface Laptop 2? I saw some EFIs but most of them are old and issues with Keyboard. Any possibility your EFI wil work with SL2? with keyboard & trackpad working?

Thank You very much in advance

w0r1d43m0n, 2025-05-19:
Same situation, trying to get it to work with Surface Laptop 4 (AMD version with NootedRed)

jlempen, 2025-05-19:
@alappatsj 
With some modifications, it could probably work for the SL2 as well. You would need to change the `AAPL,ig-platform-id` to `0000C087`, `00001659` or `00001B59` and add the `device-id` `16590000`. Then you'd probably need to change your `SystemProductName` to `MacBookPro14,1` and generate your own `SystemSerialNumber`, `MLB` and `SystemUUID` with [CorpNewt's GenSMBIOS tool](https://github.com/corpnewt/GenSMBIOS).

I can't guarantee it'll work, but I know it has been done before. Perhaps it would be easier to just search GitHub for another repository specifically made for the SL2.

jlempen, 2025-05-19:
@w0r1d43m0n 
I have no idea if NootedRed and other tools for AMD would work on the SL4, but I'm quite confident in saying that the [BigSurface driver](https://github.com/Xiashangning/BigSurface) does not support any Surface device newer than the Surface Laptop 3, Surface Book 3 and the Surface Pro 7. So I'm afraid my repo for the SL3 will not help you much, sorry.

w0r1d43m0n, 2025-05-20:
@jlempen 
yep, found that out the hard way a couple minutes ago
would it even be possible to use any of that? like with any kext or ssdt?

jlempen, 2025-05-21:
@w0r1d43m0n 

> would it even be possible to use any of that? like with any kext or ssdt?

BigSurface is pretty much the only kext specifically targeting Surface devices and only the Intel variants anyway. Perhaps there are other solutions out there, but i doubt it. Surface hardware is difficult to hackintosh.

w0r1d43m0n, 2025-05-21:
>@jlempen BigSurface is pretty much the only kext specifically targeting Surface devices and only the Intel variants anyway. Perhaps there are other solutions out there, but i doubt it. Surface hardware is difficult to hackintosh.
ah ok, thanks though!