jlempen/Surface-Laptop-3-OpenCore #17: Bluethoot cannot connect

rahmadfadhil, 2025-06-11:
I need connet the mouse in bluethoot but cannot connet, just only loading appears

jlempen, 2025-06-13:
Ah, and another thing which could fix Bluetooth in Sequoia:

Updgrade `IntelBluetoothFirmware.kext` and `IntelBTPatcher.kext` to the latest version 2.5.0 to fix Bluetooth in Sequoia 15.1 and newer. You can [download them here](https://github.com/dortania/build-repo/releases/download/IntelBluetoothFirmware-01cc180/IntelBluetoothFirmware-2.5.0-RELEASE.zip).

I'll update the whole EFI with the latest kexts this week-end.

rahmadfadhil, 2025-06-13:
> Ah, and another thing which could fix Bluetooth in Sequoia:
> 
> Updgrade `IntelBluetoothFirmware.kext` and `IntelBTPatcher.kext` to the latest version 2.5.0 to fix Bluetooth in Sequoia 15.1 and newer. You can [download them here](https://github.com/dortania/build-repo/releases/download/IntelBluetoothFirmware-01cc180/IntelBluetoothFirmware-2.5.0-RELEASE.zip).
> 
> Edit: I've updated my repo with the latest kexts, including the above kexts. Give it a try.

Thank uuu, after i try i give u information,

rahmadfadhil, 2025-06-13:
I have macOs installer on external 2 
1. Ventura 
2. sequoia 

previously could not bluetooth on sequioa, had updated kext several times all the ways I did. restart still did not work, 

after frustration I booted into ventura and bluetooth worked normally, so I booted back into sequioa and bluetooth worked normally. where is wrong? previously I restarted into sequoia did not work now it works.

jlempen, 2025-06-13:
@rahmadfadhil 
Welcome to the quirky world of hackintoshing. Sometimes shutting down, waiting a few seconds and rebooting fixes things. But anyway, Sequoia is not supported by [the OpenIntelWireless project](https://github.com/OpenIntelWireless) yet, that's one of the main reasons I recommend staying on Ventura.

If there's no important reason for you to run Sequoia, then I would suggest you install Ventura where pretty much everything works nicely.

rahmadfadhil, 2025-06-14:
Thanks a lot sir, welcome the hackintosh

We together, we will not surrender 🤣

jlempen, 2025-06-17:
@rahmadfadhil 
That's one of the limitations of the OpenIntelWireless `AirportItlwm.kext` Wifi driver. 

The only `Continuity features` supported with this driver are `Handoff` and `Universal Clipboard`. An `Instant Hotspot` from an iPhone can be recognized but the hotspot must be enabled directly on the iPhone first. The driver is also unable to connect to Hidden Networks. Source: https://openintelwireless.github.io/itlwm/FAQ.html#limitation-of-airportitlwm-kext