jlempen/Surface-Laptop-3-OpenCore #23: Complete solution to the CPU and high temperature problem

ikador, 2025-08-20:
Hello again, my dear friend,

I want to tell you something really, really interesting that even I was stunned by.
For the past month, I’ve been struggling with all kinds of patches, BIOS tweaks, and various tests trying to get the CPU and fan to behave normally so I could use the system outside of Low Power Mode—but it just didn’t work.
The situation was such that even running Cinebench Multicore outside Low Power Mode was impossible; the system would shut down due to high temperatures within the first minute of the test.

Now, after all these events that I’ve been sharing with you for a while, can you guess where the problem was? 🤣

The damn touchscreen.
The damn touchscreen!!

Yesterday, I asked you how to disable it because it’s truly useless, and to my complete surprise, all CPU and fan problems were fixed.
Even the BD-PROCHOT patch I had been using was forcibly disabled, and the system now works flawlessly, just like it does on Windows.
It easily passed a benchmark test that it previously couldn’t handle.
A single cursed driver had completely crippled the system.

Anyway, I wanted to let you know, my dear friend, in case someone else had this problem and could solve it. 😁
Thank you for your patience and support, and I wish you all the best!

ikador, 2025-08-21:
Hello dear friend,

Follow the path they mentioned in their reply to you, it’s easy to remove it.
I read your response and I think the reason you weren’t successful is this:
Instead of editing the Info.plist inside the main kext, you went and deleted one of the kexts inside the Plugins folder.
That’s a mistake.

What you actually need to do is this:
Open the main kext, which is bigsurfaceSLB3.kext (right-click it and choose Show Package Contents).
Go inside the folder — here you’ll see a file called Info.plist. Open this file with a tool such as Plist Editor, go to the path our friend mentioned, delete that specific entry, then save and exit.

Now this kext will no longer include the touchscreen driver and you can use it.
I’ll also provide you with a screenshot of the line you need to delete, dear friend.

ConwayTech-Dev, 2025-08-21:
I also made an issue in the BigSurface kext mentioning this issue, if Xiashangning ever decides to work on the kext again, perhaps it can be fixed. https://github.com/Xiashangning/BigSurface/issues/141

ikador, 2025-08-21:
Yes, it behaves the same way for me as well, and when I press restart the same thing happens.
But it still does its job and the restart completes.

jlempen, 2025-08-21:
@ikador 
Wow, this is an amazing discovery!

Did you guys and girls actually have [Xiashangning's IPTSDaemon](https://github.com/jlempen/Surface-Laptop-3-OpenCore#installing-the-iptsdaemon-to-enable-the-touchscreen) installed on your SL3? Perhaps the overheating and throttling issues due to the touchscreen could be resolved by installing the IPTSDaemon.

I'm asking this because I don't recall having such issues with my 15" SL3, but I admit that I don't really use this machine on a daily basis. I'm only toying around with my SL3. But I actually used it to compile dozens of plists when I was trying to fix the `BigSurface.kext` and I never heard the fan spin up and my temperatures stayed pretty much constantly below 50 degrees centigrade. The `IPTSDaemon` is installed on my machine and the low power mode is set to `always on`, though.

I'm very busy at work at the moment and don't have the SL3 with me, but it would be interesting to see what happens on your machines when the touchscreen is enabled and the `IPTSDaemon` is installed.
If someone confirms that the overheating / throttling issues persist even with the daemon installed, then I'd be more than happy to disable the touchscreen in my EFI and update the readme accordingly.

I agree that the touchscreen is an absolute nonesense on a laptop such as the SL3.

jlempen, 2025-08-21:
@ikador 

> Finally, just a couple of days ago, since I’m using Sequoia, I decided to disable two things that were putting extra load on the system: Bluetooth and the touchscreen.
> Bluetooth didn’t make a huge difference, but it helped a little, especially in Sequoia because it produces a lot of errors.

The commits https://github.com/jlempen/Surface-Laptop-3-OpenCore/commit/11a6e8185aeab173d9bd0b8172676072d6214073, https://github.com/jlempen/Surface-Laptop-3-OpenCore/commit/c12a67ac2ef833a556f3256ea91f61b1432d6ae6 and https://github.com/jlempen/Surface-Laptop-3-OpenCore/commit/97dbb1010295debe05313c0f70eb366d155cddde from last month actually fixed many bluetooth issues on Sequoia.

ConwayTech-Dev, 2025-08-22:
> Did you guys and girls actually have [Xiashangning's IPTSDaemon](https://github.com/jlempen/Surface-Laptop-3-OpenCore#installing-the-iptsdaemon-to-enable-the-touchscreen) installed on your SL3? [...]
Hi @jlempen. I was experiencing the issue while having IPTSDaemon installed, actually. Seems like having it installed doesn't change anything -- I've had it installed since day 1.

ConwayTech-Dev, 2025-08-24:
Nothing works. The SL3 panics, and despite the boot arguments, restarts itself. I'm guessing there's some kind of obvious error in my EFI -- I've tried both @jlempen and @ikador's configs, and neither worked. Could one of you look through my EFI and try to find the issue? I've tried everything, including restarting around 130-140 times.

GitHub won't let me upload my EFI, since it's too large, so I put it on buzzheavier. https://buzzheavier.com/euzn6fnq7672

Thanks in advance! This is a really frustrating situation.

ikador, 2025-08-24:
@ConwayTech-Dev 

One thing that usually helps me in these situations is using ProperTree and creating a snapshot.
You might want to try it and see if it solves your problem.
By the way, in the screenshot you shared, two of your main kexts showed errors — Lilu and WhateverGreen.
If you’ve updated them, take another snapshot, because maybe the kext versions got messed up.

ConwayTech-Dev, 2025-08-24:
I think the reason my Hackintosh isn't working is actually because something went wrong with my APFS partitions or partitions in general. 80% of the errors are about "apfs_keybag_init" fails or "apfs_trim_to_prev_fsize" errors. And that would also explain why my EFI looks perfect -- those are macOS errors, not OpenCore errors. But it's weird, because everything I've done so far has been in Linux Mint:
- Create /mnt/efi
- Mount the EFI partition there
- Make changes
- Sync
- Unmount EFI

So I'm pretty worried about my SSD being corrupted.

ConwayTech-Dev, 2025-08-25:
Perhaps because you two are using MOD-OC? I can test, but other than that and what @jlempen said, it should definitely work.

ConwayTech-Dev, 2025-08-25:
> [@ConwayTech-Dev](https://github.com/ConwayTech-Dev) I know it sounds strange that this actually works, but I only quoted Xiashangning's reply in a two years old thread in his [BigSurface](https://github.com/Xiashangning/BigSurface) repo on how to disable the touchscreen [here](https://github.com/Xiashangning/BigSurface/issues/102#issuecomment-1505505928).

I saw that too, actually, when looking through the BigSurface issues to see if there was anything interesting, haha.

ikador, 2025-08-25:
@ConwayTech-Dev 

In the beginning, I was using the MOD version of OpenCore, but I’ve updated to the official release a while ago. The version I’m using now is OpenCore Release 1.0.5.

ConwayTech-Dev, 2025-08-25:
Thanks. I got it working by fixing my kext order, for some reason I was loading BigSurfaceSLB3.kext at the wrong time. Now I'm going to test whether it fixes thermals, hoping for the best!

ConwayTech-Dev, 2025-08-26:
Yup! The laptop is booting, my touchscreen is disabled, and everything works great.

For some reason, using one-key-hidpi didn't allow me to get HiDPI resolutions over 1280x854, which is a bit too large for my liking. I wanted 1440x960, but I'm guessing that macOS thinks it's not close enough to 2x the actual panel resolution and refuses to render it at 2x and downscale. I was using BetterDisplay Flexible Scaling before, so I guess I'll have to go back to that, which uses an alternative method for HiDPI.

As for the small issues, sounds good, we'll just have to work on that when Tahoe releases.

ConwayTech-Dev, 2025-08-30:
@jlempen, sorry for the late reply. Your suggestion worked perfecrly, thanks! All there is now, I suppose, is just to wait for Tahoe and make a good EFI when it releases. Talk to you then!

jlempen, 2025-09-01:
@ConwayTech-Dev 
Ok, cool you got it working now!

As Tahoe is quite broken for everyone at the moment (broken audio and bluetooth causing random kernel panics), I'll wait for fixes to `AppleALC` and `IntelBTPatcher` before giving it a spin. I'm also very busy at work until mid-October, so don't expect anything before November...

I'll disable the touchscreen in my EFI, though, as this seems to fix quite a few things for everybody.

jlempen, 2025-09-01:
Fixed with https://github.com/jlempen/Surface-Laptop-3-OpenCore/commit/1745381a900c55adf230c7bbfec65a31d30711fa

Feel free to reopen this issue if disabling the touchscreen still doesn't fix your issues with throttling and overheating

amethyst8118, 2025-10-15:
> Fixed with [1745381](https://github.com/jlempen/Surface-Laptop-3-OpenCore/commit/1745381a900c55adf230c7bbfec65a31d30711fa)
> 
> Feel free to reopen this issue if disabling the touchscreen still doesn't fix your issues with throttling and overheating

It does fix it to an extent, but anything resource intensive will throttle the CPU to 400Mhz. The only way to get reliable performance is disabling BD-PROCHOT, and again as I have said before, disabling it is fine, the CPU will still thermal throttle or shutdown if temps reach a certain threshold. 

After all, the actual MacBook gets just as hot if not more on the same silicon.

amethyst8118, 2025-10-16:
> Thank you for your input, [@amethyst8118](https://github.com/amethyst8118)
> 
> Just to be sure, please confirm that you've disabled the touchscreen, that your firmware was downgraded to 13.101.140.0 and that you're using `BigSurfaceSLB3.kext` and not the failsafe `BigSurface.kext`, right?
> 
> If so, I'll add DisableBDPROCHOT.kext to my EFI, perhaps disabled at first because I'm afraid it could overheat the SL3 during the installation process.

Yep, on both MacOS Ventura and Tahoe (doing some finishing touches on that now, glad to know BigSurface works flawlessly even on the last x86 MacOS), and I too think just adding a note on readme with a disclaimer would be nice.

jlempen, 2025-10-16:
@amethyst8118 DisableBDPROCHOT.efi added to the EFI with https://github.com/jlempen/Surface-Laptop-3-OpenCore/commit/9b735113b12da9855e151ed7367125a66bf91516

amethyst8118, 2025-10-16:
> [@amethyst8118](https://github.com/amethyst8118) If BigSurface works flawlessly, there's probably not much else needed for Tahoe to run on the SL3. Did you find a way to fix sound yet?

need to partially disable SIP and install AppleHDA.kext, going to do that now, I got locked out of my OS as Tahoe forces filevault or something every reboot. There's a fix for that I found from Reddit: 

> use apfs_aligned.efi from ocbinarydata on uefi->drivers and disable enablejumpstart on uefi->apfs

amethyst8118, 2025-10-19:
> [@amethyst8118](https://github.com/amethyst8118) I tinkered with Tahoe 26.0.1 tonight and it installs and already runs quite nicely!
> 
> I updated the `USBMap.kext` for Tahoe compatibility and made promising progress on Bluetooth. Have a look at the stuff I merged into my EFI today.
> 
> I haven't bothered to add the existing fixes for FileVault yet as I don't use encryption on this laptop. I'll work on audio next.

Any updates on Audio ? I used VoodooHDA.kext with SIP partially disabled.