jlempen/Surface-Laptop-3-OpenCore #25: Old SL3 firmware needed.

BrunetteBird57, 2025-08-31:
Hello, this is not an issue with running MacOS on the SL3 at all, so please feel free to close/remove this issue if you deem necessary to do so, as I do not want to be polluting or irritating anyone. 

I cannot seem to find any place else to ask this, so I might aswell - is there any repository/website/person/whatever, where I could get .cab files of the oldest versions of the SL3 Intel ver firmware? While the one posted in the rep is great and works (13.101), I'd like an older one, specifically for the purpose of trying to reverse plundervolt. Like I said, since this is unrelated, feel free to remove it if it's irritating in any shape or form. Otherwise, if anyone has any info regarding this, I'd highly appreciate it, especially info regarding versions from October and November of 2019. I know there are .msi downloads, however, they're unworkable by fwupdmgr even through unpacking the .bin files from the .msi file, nor have I found a way to somehow convert any of those files to .cab, not that there isn't any - it's just probably my lack of knowledge. Like I said, I'd greatly appreciate any help guidance or info regarding this topic.

BrunetteBird57, 2025-08-31:
Hello, I hope you're good aswell. I've tried out the git that you've linked to me about a year ago actually. I've had many issues with the process, mostly getting stuck on the repack.sh script that would simply not work, emitting tons of different errors. When I'd eventually find a fix for the one that would occur at the time, another one would show up in its place. I also tried some Windows .cab file repackers, but unfortunately the files created by that software would not be read successfully by fwupdmgr, stating the archive is corrupted. I tried the process both on Ubuntu (I believe version LTS 21?) and some other, supposedly more minimal distro that I cannot now remember the name of. I'd of course update msitools, gcab and dos2unix using apt, so unless the version of the linux distro I was using somehow impacted one of those specific tools (which I'd not be able to say, I'm not really that well acknowledged with Linux), I'm not sure what I did wrong. I gave up stuck on my attempts to edit the repack.sh script to somehow make it work. I guess in a moment of free time I'll try my hand at the solution mentioned in that git again. Thanks for your response.

jlempen, 2025-09-01:
@BrunetteBird57 
You're not irritating at all :-) There is not much information available on the net about downgrading the firmware of Surface devices. As @ConwayTech-Dev suggested, the only helpful resource I was able to find is in [the linux-surface repo](https://github.com/linux-surface/surface-uefi-firmware), but many people have issues with this script. Please have a look at the issues section of that repo as well, there is some really helpful stuff there.

Since I wrote [my howto for downgrading the firmware of the SL3](https://github.com/jlempen/Surface-Laptop-3-OpenCore#downgrading-the-uefi-firmware) in my repo, most people reported a success, but a few other users had problems getting this method to work. It seems to all boil down to my zipped firmware files being too old and the fwupd tool therefore failing. I can't find the exact issue thread in my repo, but I recall that one user found that downloading the msi file again from Softpedia and extracting it again using the instructions in the linux-surface/surface-uefi-firmware repo solved the issue.

Also have a look at an [older post I wrote last year](https://github.com/Xiashangning/BigSurface/issues/79#issuecomment-2208484390).

And then there's some interesting findings by @pnxl in his [instructions on how to prevent Windows from updating the firmware to the latest version again](https://github.com/jlempen/Surface-Laptop-3-OpenCore#running-windows-without-updating-the-firmware). He seems to suggest that installing old driver packs for Surface devices also downgrades the firmware to the version of that very driver pack. You can find older driver packs for Surface devices on Softpedia.